eukatree/Bambu_CustomGCode #10: wrong positions for A1 mini

Hillbilly-Phil, 2024-07-09:
Found some issues on the A1 mini code and patched them as good as I could. Runs smooth for me now:

-flushing takes place on wrong side of the bed --> added the right x coordinate
-extruder doesn't move far enough to the right to succesfully cut the filament --> added a slow move to cut the filament

Instructions for changing the filament with this modified gcode:

;===== Instructions: =====
; when print has paused, go into control
; set the nozzel temp to the temp you were printing with
; unload the filament by pushing on the upper extruder-button 
; load the new filament by pushing on the lower extruder-button 
; resume the print (flushing will happen next, flushing volumes can be set in Bambu Studio as if using an AMS)

[;===Bambu Lab A1 mini 0.4 GITHUB NO.txt](https://github.com/user-attachments/files/16149302/Bambu.Lab.A1.mini.0.4.GITHUB.NO.txt)

pakonambawan, 2024-08-01:
i polished it right here. 
now after it cuts and goes back to it's place to poop, now it retracts the filament so just pull it out then just place another filament back in. 

[A1 mini MMS auto pause.txt](https://github.com/user-attachments/files/16457326/A1.mini.MMS.auto.pause.txt)

khristangrey, 2024-08-08:
Will this work on the A1?

pakonambawan, 2024-08-08:
> Will this work on the A1?

I haven't tried yet but I think no.  A1 mini have a smaller build plate so the gcode positions to where the cutter and poop ejector are different.

khristangrey, 2024-08-08:
Got it. I want it to work on the A1. Can you point on what line I should change for the cutter positions and cutter? I have no experience on working on gcodes.

Hillbilly-Phil, 2024-08-08:
Take a look at these lines:

[…]
G1 X0 Y90 F18000
G1 X-13.5 F9000
G1 E-13.5 F900

; pause for user to load and press resume
M400 U1
[…]

Meaning:
G1 = linear move to Position X Y Z with speed F
G1 E-… = Retract Filament by … mm
M400 U1 = Pauses the print till user resumes

So copy paste the whole code open in an text editor and play Around with the Coordinates X Y Z till you are satisfied with the positions. 
XYZ corespond to the xyz in the shown in the slicer. 
cheers

Hillbilly-Phil, 2024-08-08:
Have a updated version with the toolbars cutting and retracting the filament wich I can upload as soon as I understand how to correctly commit on GitHub 😅 … and not just in the comment section.

khristangrey, 2024-08-08:
I got it to work! I got the coordinates from the original gcode of Bambu A1. Thank you so much! 
[A1 No AMS.txt](https://github.com/user-attachments/files/16541185/A1.No.AMS.txt)

Dennis-Q, 2024-08-21:
@Hillbilly-Phil  @pakonambawan and @eukatree  thank you for sharing all of this.

The instructions in the gcode was really useful (to not use the 'unload' / 'load' option). Also the finetuning done look great. I took the time to use your modifications and combine them with the latest version that is supplied with the latest Bambu Studio.

I have put your names in the gcode file and README.md since you are the contributors to the file as well.
The code on the github has been tested many times and works great for me with the latest Bambu Studio and A1 Mini firmware. Still, I am open to any suggestions for improvements if you have any.

My approach with the gcode file is to stay with the original code as much as possible and only change what is really mandatory to change (e.g. the gcode that causes the printer to crash). Also I tried to document what has been removed, added or moved, within the file. That could help later to understand the changes and why they were done. :)

You can find the code (for the A1 mini) here:
https://github.com/Dennis-Q/bambu

wundergeu, 2024-08-25:
Thanks for your input concerning the necessary changes for the A1 (mini).

Out of curiosity:
Could I also just edit the 'Change Filament G-Code' section to a single line of 'M400 U1' and then load/unload the filament manually with the menu buttons on the display?

Dennis-Q, 2024-08-25:
> Thanks for your input concerning the necessary changes for the A1 (mini).
> 
> Out of curiosity: Could I also just edit the 'Change Filament G-Code' section to a single line of 'M400 U1' and then load/unload the filament manually with the menu buttons on the display?

Nope, that's what I thought but things will not work as expected. One of the reasons is that an important part of the code is only executed when the AMS is connected and responding, which it is not.

Unload en load should not be used. Use Control and Extruder buttons as in the gcode instructions...

wundergeu, 2024-08-25:
> Nope, that's what I thought but things will not work as expected. One of the reasons is that an important part of the code is only executed when the AMS is connected and responding, which it is not.
> 
> Unload en load should not be used. Use Control, Raise Temp and Extruder buttons as in the gcode instructions...

Alright, thanks for the quick response.
Gonna adapt the code for the A1. Your comments in the gcode make it way more understandable, so thanks for that.

Joseluisuuj, 2024-08-31:
> ¡Lo hice funcionar! Obtuve las coordenadas del código G original de Bambu A1. ¡Muchas gracias! [A1 No AMS.txt](https://github.com/user-attachments/files/16541185/A1.No.AMS.txt)
hola, me he dado cuenta que la pausa la hace bien pero cuando se desplaza a la esquina a cortar el filamento no llega a cortarlo ya que no la presiona lo suficiende.
seria cosa de modificarlo para que avance mas y por ello la presione un poco mas y corte el filamento.
por lo demas el gcode para la bambulab A1 va bien

Joseluisuuj, 2024-09-01:
> > ¡Lo hice funcionar! Obtuve las coordenadas del código G original de Bambu A1. ¡Muchas gracias! [A1 No AMS.txt](https://github.com/user-attachments/files/16541185/A1.No.AMS.txt)

hola, me he dado cuenta que la pausa la hace bien pero cuando se desplaza a la esquina a cortar el filamento no llega a cortarlo ya que no la presiona lo suficiende.
seria cosa de modificarlo para que avance mas y por ello la presione un poco mas y corte el filamento.
por lo demas el gcode para la bambulab A1 va bien

Joseluisuuj, 2024-09-01:
#

> ¡Lo hice funcionar! Obtuve las coordenadas del código G original de Bambu A1. ¡Muchas gracias! [A1 No AMS.txt](https://github.com/user-attachments/files/16541185/A1.No.AMS.txt)

hola, me he dado cuenta que la pausa la hace bien pero cuando se desplaza a la esquina a cortar el filamento no llega a cortarlo ya que no la presiona lo suficiende.
seria cosa de modificarlo para que avance mas y por ello la presione un poco mas y corte el filamento.
por lo demas el gcode para la bambulab A1 va bien

wundergeu, 2024-09-01:
> english: Hello, I have realized that the pause works well but when you move to the corner to cut the filament, it does not cut it because you do not press it hard enough.

Same for me. It doesn't seem to be "strong" enough to push the cutting lever. I don't know why it seems to work for others, but not for some.
I deleted the gcode for the cutting motion and just use the menu buttons to unload/load when it is paused.

Joseluisuuj, 2024-09-02:
@khristangrey  Could you modify the gcode so that it would push the lever well and cut the filament?  thanks in advance

khristangrey, 2024-09-03:
@Joseluisuuj Yes. I have not tried it again. But you are correct, it does not cut the filament. Will adjust it and will repost it here.

Joseluisuuj, 2024-09-04:
Thank you very much, since that is the only thing I can print multicolor with

Joseluisuuj, 2024-09-12:
@khristangrey you ajust the gcode??

khristangrey, 2024-09-12:
So I tried to adjust the cutting motion but it won't work. The maximum movement I can do is 283 in this line:

[…]
G1 X256 F18000
G1 X283 F500 //here
G1 X256 F500
[…]

What happens if I add more, it grinds. Don't know why I can't make it to cut. For now, I would be manually pushing the cutting lever to cut the filament.

mattmichaels, 2024-10-01:
perhaps these should be posted to the main repo?

i tried to print the repo last night and noticed the coordinates were not matching due to a different printer. would imagine others are having the same - didn't know the solution was hidden in open issue comments, haha.

1100101, 2024-10-02:
> So I tried to adjust the cutting motion but it won't work. The maximum movement I can do is 283 in this line:
>
> […] G1 X256 F18000 G1 X283 F500 //here G1 X256 F500 […]
>
> What happens if I add more, it [...]

1. A1 mini users, be careful: the X movement numbers are for a A1, not the mini! This could damage the printer
2. Instead of increasing the X movement, I lowered the speed from 500 to 300. Doesn't work with all filaments, but better than before.

mattmichaels, 2024-10-02:
i printed this on my A1 yesterday and had no cutting issues. 

once the pause started, i entered the control section and raised the temp of the extruder and extruded (up) to unload the current filament and then extruded (button down) to load the new filament. 

worked like a charm

JimmyJo3D, 2024-11-08:
does anyone know if this gcode is the same one used here for a1?
https://makerworld.com/it/models/631637#profileId-556604

BencsikDani, 2025-12-09:
> Hey Guys,
> 
> As I haven't really found a complete working "NoAMS + AutoUnload" code for the **A1 mini** that works, I took the original code from the Bambu Studio System presets, and I created my own from scratch. Of course I had lots of sources to get proper knowledge for it, like this original repo, other repos, and I also sprinkled with some AI magic.
> 
> I formatted & commented the original code, removed the AMS-related commands, and added Cut & Unload things.
> 
> With this implementation, the filament is really automatically unloaded, so when print is Paused, you just have to pull the filament out, feed the next one and hit Resume.
> 
> I hope it helps!
> 
> [a1_mini_change_filament_noams_autounload.txt](https://github.com/user-attachments/files/23597004/a1_mini_change_filament_noams_autounload.txt)

Hey again,

After some weeks I finally realized how terrible my code was.
I am still figuring out some things, but until next time here is v2, that actually works.

[a1_mini_change_filament_noams_autounload_v2.txt](https://github.com/user-attachments/files/24043972/a1_mini_change_filament_noams_autounload_v2.txt)

Currently the flush amount during filament change is hardcoded. I found ~70-80 mm to be good enough. But feel free to change it in line 77.

I hope it helps!