eukatree/Bambu_CustomGCode #21: Please: Need for A1 (non mini)

bigjoncoop, 2025-01-31:
Can anyone confirm that this works for the full size A1 (non mini)?

Or if possible could someone modify to work?

Thanks

smbunn, 2025-02-06:
I loaded this onto my full size A1.  Behaviour is not what you would expect.  It does pause between color (filament) changes but moves erraticaly, then moves plate to the very front and starts oozing filament  besides the rubber wiper block down into the solt into the rails,  I had to catch it on a bit of paper.  Then it goes to the front again and stops and 'says paused'

cotestatnt, 2025-02-07:
This is my actual gcode for change filaments with the open source project [3D Chamaleon](https://www.3dchameleon.com/) on a Bambulab A1.
It is a mix between the original Bambulab gcode and other sources.

In order to perform a manual color change, you can uncomment the command M400 and comment the dwell command G4.
```gcode
;G4 P5000 ;wait for stepper movement complete
M400 U1	; pause for user to load and press resume
```

You can also delete the previous section for the necessary color selection using the 3D Chameleon.

```gcode
{if next_extruder==0}
    G4 P600 ; dwell for .6 seconds - adjust this to match your machines single pulse time
{endif}
{if next_extruder==1}
    G4 P1100 ; dwell for 1.1 seconds - adjust this to match your machines two pulse time
{endif}
{if next_extruder==2}
    G4 P1600 ; dwell for 1.6 seconds - adjust this to match your machines three pulse time
{endif}
{if next_extruder==3}
    G4 P2100 ; dwell for 2.1 seconds - adjust this to match your machines four pulse time
{endif}
```

My complete gcode (used with 3D Chamaleon custom firmware):

```gcode
;===== machine: A1 =========================
;===== date: 20240830 =======================

;previous_extruder: [previous_extruder]
;toolchange_count: [toolchange_count]
;flush_length_1: [flush_length_1]
;flush_length_2: [flush_length_2]

M1007 S0 ; turn off mass estimation
G392 S0  ; turn off clog detect
;M620 S[next_extruder]A
M204 S9000 ; set starting acceleration

{if toolchange_count > 1}
G17
G2 Z{max_layer_z + 0.4} I0.86 J0.86 P1 F10000 ; spiral lift a little from second lift
{endif}
G1 Z{max_layer_z + 3.0} F1200

M400
M106 P1 S0
M106 P2 S0

; {if old_filament_temp > 142 && next_extruder < 255}
; M104 S[old_filament_temp]
; {endif}

{if toolchange_count>0}
; G1 X267 F18000
; M620.1 E F[old_filament_e_feedrate] T{nozzle_temperature_range_high[previous_extruder]}
; M620.10 A0 F[old_filament_e_feedrate]
; T[next_extruder]
; M620.1 E F[new_filament_e_feedrate] T{nozzle_temperature_range_high[next_extruder]}
; M620.10 A1 F[new_filament_e_feedrate] L[flush_length] H[nozzle_diameter] T[nozzle_temperature_range_high]

; M73 P99 R99

G0 E-5 F500 ; retract a bit, adjust this to tune waste

M17 X1.0 ; increase X motor current
G1 X279 F18000; cut filament
[... 230 lines cut ...]
  G1 X-48.2 F3000
  G1 X-38.2 F18000 ;wipe and shake
  G1 X-48.2 F3000
  G1 X-38.2 F12000 ;wipe and shake
  G1 X-48.2 F3000
  M400
  M106 P1 S0 
  

M623

; M621 S[next_extruder]A

; as there is no AMS, these next three lines only serve to hide T[next_extruder]
; if this was not included, the T[next_extruder] command is input after this
; code and will cause the system to hang as the toolchange command searches
; for the AMS
M620 S[next_extruder]A
T[next_extruder]
M621 S[next_extruder]A

G392 S0

M1007 S1
```

otede, 2025-08-29:
@cotestatnt Hi! Could you please share the de-cameleonized gcode as well? ;)